cmusatyalab/steeleagle #24: requirements.txt is no longer generated and included in the .ms zip file

teiszler, 2024-04-03:
In hermes, we used pipreqs to generate a requirements.txt file that was then packed into the .ms (zip) file. This no longer appears to be happening in Compiler.java.

```
            # Generate requirements.txt inside the python directory
            os.system("cd ./python; pipreqs . --force")
            
            # Add requirements.txt to the zip, under the new directory
            zf.write("./python/requirements.txt", arcname=f"./requirements.txt")
```

teiszler, 2024-04-03:
https://github.com/cmusatyalab/steeleagle/blob/70351bf1d065612cd49bc87014527193af175c0e/droneDSL/compile/src/main/java/org/droneDSL/compile/Compiler.java#L85

xianglic, 2024-04-04:
Pushed the new commit for adding requirements.txt

teiszler, 2024-04-04:
Fixed in [7fc3189](https://github.com/cmusatyalab/steeleagle/commit/7fc3189c3c788debc660ac2dba2993a210bd6dae).

closed 2024-04-04 by teiszler