cmusatyalab/steeleagle #30: Driver module does not connect to the olympe server properly

xianglic, 2024-09-27:
after restarting the container, the driver module will not connect to the olympe server for 7-8 min. The restarted container could still ping the onion.

teiszler, 2024-10-10:
This seems to have something to do with the docker network being torn down and then brought back up with `docker compose down`. If you just bring down the driver container by itself, and leave the network as is, then the driver can still talk to the drone over olympe.

xianglic, 2024-11-27:
Notice: do not use docker compose down <service> for bringing down the certain service. This cmd has some backward compatibility issue that might not be able to leave the network even if you choose to bring down certain service. Use docker compose stop <service> docker compose rm <service> and docker compose up <service>.

xianglic, 2024-11-27:
Here is the reference of the issue of docker compose down : https://stackoverflow.com/questions/41450741/why-doesnt-docker-compose-down-take-an-optional-service-argument

closed 2025-04-03 by mihirbala